mikakaraila/node-red-contrib-opcua #795: Multiple Servers Error

VitorCaldana, 2025-03-29:
I was running an application on version 0.2.331 with 4 different servers Ok

I´ve got a new laptop and installed the newer version 0.2.339 and 2 of the servers won´t run. They give me an error:

populate_data_type_manager_103:757  -  Cannot process any more dataTypeDictionary

Can anyone help me with that?

If not, is it possible to installed the older 0.2.331 version? where can I get ir?

mikakaraila, 2025-03-31:
Try this from command line:
npm install node-red-contrib-opcua@0.2.331

VitorCaldana, 2025-03-31:
I did get the version 0.2.311 running, however the issue remains. 
I´m looking into issues with Windows 11 now, since the older versions ran on Windows 10.

mikakaraila, 2025-04-01:
I will try to find some time during this week to look these issues....

VitorCaldana, 2025-04-10:
I got the follwing error when updating the flows with multiple servers. Seems it is something with the certificates.
Does not matter if I´m at Windows 10, 11 or which version of node-red-contrib-opc-ua is installed.

[image]

The service crashes and I have to manually restart Node-RED in CMD.